OctoPrint/OctoPrint-AutomaticShutdown #1: Sidebar location

MoonshineSG, 2015-10-12:
can this be moved ABOVE the Files list ? It would be much more visible...

nicanor-romero, 2015-10-13:
Hi @MoonshineSG ! I'm afraid that is not possible. As stated here http://docs.octoprint.org/en/master/plugins/mixins.html#templateplugin plugins can only add elements beneath the existing sections.

Salandora, 2015-10-13:
Hi @MoonshineSG
you could try this: http://docs.octoprint.org/en/master/configuration/config_yaml.html#appearance

As I understand it right you can simply define the order the items appear.

MoonshineSG, 2015-10-13:
Never mind. The plugin doesn't really work so I removed it.

nicanor-romero, 2015-10-13:
@MoonshineSG Do you mind explaining why it isn't working for you?

MoonshineSG, 2015-10-14:
Apologies!!! It was entirely my fault. All I had to do was RTFM.  I am not using "systemShutdownCommand". This is not a plugin issue.

closed 2015-10-14 by MoonshineSG