OctoPrint/OctoPrint-AutomaticShutdown #3: Feature Requests

hongkongkiwi, 2016-05-04:
Great plugin thanks!

If possible I would like to request the following features:
- Setting to default automatic shutdown to true
- Ability to set the action that's performed (instead of shutdown), as I want to have a different system shutdown to just turning off the printer

malebuffy, 2016-11-06:
I also want a custom command since I am using gpio pin 23 (pin 4 from witingPI) to power on/off my printer.

totalitarian, 2017-01-08:
custom command would be great

kantlivelong, 2017-01-26:
@hongkongkiwi @malebuffy: Perhaps this plugin might interest you http://plugins.octoprint.org/plugins/psucontrol

As for the default value of the checkbox, would it make sense to have an option to remember the state?

malebuffy, 2017-01-27:
thanks for the link! yes would make sense!

kantlivelong, 2017-02-15:
Added option to remember last value of checkbox.

Will be in upcoming release.

closed 2017-02-15 by kantlivelong