OctoPrint/OctoPrint-AutomaticShutdown #4: Change timeout, so camera is given more time to make a picture

fraternl, 2016-06-07:
I'm using your plugin for automatic shutdown as I'm using a plugin that will mail me a picture of the finished print.
The camera on my Raspberry Pi is unable to make a decent picture without enough lighting.
The lighting of the printer is hardwired to the power supply of the printer.

When the printer shuts down after printing it will not wait long enough for the picture to be taken.

nicanor-romero, 2016-06-07:
Hi @fraternl . Yes, that should do the trick :)

Helisinus, 2016-10-23:
Where can i found this init.py ?
i need to change this value too

fraternl, 2017-01-24:
The file name is not init.py
Github is doing that

`find -name __init__.py | grep automaticshutdown
`

kantlivelong, 2017-02-06:
Would anyone be willing to test out a new devel version that fixes(hopefully) this? I currently do not have a camera.

kantlivelong, 2017-02-15:
Fixed in upcoming release.

closed 2017-02-15 by kantlivelong