OctoPrint/OctoPrint-AutomaticShutdown #5: Access to checkbox when not logged in?

chunkysteveo, 2016-06-09:
When I access my Octoprint server as a standard user (not logged in) the tickbox to shut down the system is visible. Will this allow anyone (who has access) to shut down your Raspberry Pi?

markwal, 2016-06-10:
Which checkbox (where do you see it)?  In my set up, that command is only on the System menu which only appears when I'm logged in.

chunkysteveo, 2016-06-10:
When I access the web frontend of OctoPrint as a non logged in user I see the "Shutdown after print finishes" checkbox on the left hand nav column under the Files box (as per the screenshot of the plugin readme). I thought I wouldn't be able to see that unless I was logged in? Have not tested it, but was able to check/uncheck it as a non-logged in user.

markwal, 2016-06-10:
Sorry about that.  I had lost context when going through the notifications. 

Looking at the code, it looks like it needs to implement the is_api_adminonly method: http://docs.octoprint.org/en/master/plugins/mixins.html?highlight=admin#octoprint.plugin.SimpleApiPlugin.is_api_adminonly

Though the js then needs to fail gracefully when it can't get the current value from the api and recover after login.

Is @nicanor-romero still maintaining this plugin?

nicanor-romero, 2016-06-10:
Hi guys. I'll try to fix this over the weekend ;)

chunkysteveo, 2016-06-10:
Ahh great stuff, quickly spotted guys!

kantlivelong, 2017-02-15:
Fixed in upcoming release.

closed 2017-02-15 by kantlivelong