OctoPrint/OctoPrint-AutomaticShutdown #6: Check box flag disappear

alebrescia, 2016-08-11:
if i put the flag to automatic shutdown and then close the browser or the tab, when i try reconnect to the octoprint web interface the flag on the automatic shutdown checkbox was disappear... 
i think that's only a "visual" problem because the auto shutdown works fine.....
sry for my bad english.... i hope you can understand..... ^_^

Helisinus, 2016-10-21:
Hi, same issue here. is it possibe to "fix" the visability of the checkbox?

totalitarian, 2017-01-12:
Same issue too. Please fix this :)

foosel, 2017-01-13:
@nicanor-romero do you still maintain this plugin or should it be considered abandoned? If so I'll look if I can find someone who can adopt it.

nicanor-romero, 2017-01-17:
@foosel Hi Gina! I'm afraid I don't have time to maintain this plugin at the moment. So feel free to find someone who can :)

foosel, 2017-01-17:
@nicanor-romero ok, will do :) I assume that also applies to any other plugins by you? 

Hope all is well on your end!

nicanor-romero, 2017-01-17:
@foosel Yes, the other plugins as well.

kantlivelong, 2017-01-23:
@alebrescia would you be willing to test out a fix from my repo?

https://github.com/kantlivelong/OctoPrint-AutomaticShutdown

totalitarian, 2017-01-24:
@kantlivelong I will, how to install it? Just download the files and copy them onto the SDcard?

kantlivelong, 2017-01-24:
@totalitarian

Assuming you are running Octoprint on linux:

1. Uninstall the plugin from plugin manager.
2. From the URL provided download and copy octoprint_automaticshutdown to the octoprint local plugins directory (in my case /home/pi/.octoprint/plugins/)
3. Rename the copied directory octoprint_automaticshutdown to automaticshutdown.
4. Restart Octoprint

foosel, 2017-01-24:
Uninstalling and just copy-pasting the URL of the zip download:

[image]

into Plugin Manager 

[image]

should also work.

totalitarian, 2017-01-24:
Thanks, it doesn't seem to work. If I check the box, then reload the page, the box is no longer checked

kantlivelong, 2017-01-25:
Whoops. I solved syncing of the checkbox state across connected clients but neglected to get the state when first connecting. I was able to test it on my side and it appears to be okay now.

Could you try again with the same URL?

totalitarian, 2017-01-25:
@kantlivelong Perfect! That you so much for fixing this :+1

closed 2017-01-26 by foosel