OctoPrint/OctoPrint-AutomaticShutdown #9: global name 'make_response' is not defined

mikecarr, 2017-02-17:
continuing to see this error in logs 

2017-02-16 21:40:53,370 - octoprint - ERROR - Exception on /api/plugin/automaticshutdown [POST]
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/server/util/flask.py", line 1110, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.1-py2.7.egg/octoprint/server/api/__init__.py", line 97, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_automaticshutdown/__init__.py", line 56, in on_api_command
    return make_response("Insufficient rights", 403)
NameError: global name 'make_response' is not defined

kantlivelong, 2017-02-17:
What version of OctoPrint are you running?

mikecarr, 2017-02-17:
v1.3.1

kantlivelong, 2017-02-17:
Could you try this fix?

1. Uninstall plugin
2. Install plugin using UR: https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/archive/issue-9.zip
3. Test

mikecarr, 2017-02-17:
>    OctoPrint-AutomaticShutdown/archive/issue-9.zip

I fixed it locally but will try this out and let you know.

mikecarr, 2017-02-17:
How can I test this out?

kantlivelong, 2017-02-17:
That's the weird thing. The plugin wont send API calls if **access control** is enabled and your not authenticated. It's primarily to prevent malicious intent.

How often was the error occurring?

mikecarr, 2017-02-17:
Can't really tell, at random times, the Cura error also occurs at the same
time

2017-02-16 19:19:10,543 - octoprint.plugins.cura - INFO - Path to
CuraEngine has not been configured yet or does not exist (currently set to
None), Cura will not be selectable for slicing
2017-02-16 19:19:10,774 - octoprint - ERROR - Exception on
/api/plugin/automaticshutdown [POST]

kantlivelong, 2017-02-17:
Hmm. Going to look into this further as I may not even need user_permission check.

kantlivelong, 2017-02-22:
Fixed in v0.1.3

closed 2017-02-22 by kantlivelong

mikecarr, 2017-02-22:
confirmed fixed, thanks for the quick response.