OctoPrint/OctoPrint-AutomaticShutdown #10: AutomaticShutdown does not shutdown

zegreatclan, 2017-02-25:
Hi ! I'm currently trying to use your plugin to shutdown my printer after the print but it seems that something is preventing it.

I tool a look at the code, it looks like Octoprint is still reporting state as printing while the printing is effectively finished :

2017-02-25 10:33:22,795 - octoprint.util.comm - INFO - Finished in 5645.460 s.
2017-02-25 10:33:22,809 - octoprint.plugins.automaticshutdown - INFO - Printer is currently printing/paused. No shutdown action will be preformed.

Octoprint v1.3.1
OctoPrint-AutomaticShutdown v0.1.3
Printer Anet A8 (Repetier firmware)

Process : start printing, check automatic shutdown, wait an infinite time, come back when print finishes, and ... nothing :)

If you need any additional info tell me !

kantlivelong, 2017-02-25:
I'll take a look.

Are you using a webcam/timelapse?

zegreatclan, 2017-02-25:
Yes I use webcam, but timelapse disabled (for now).  I've tried two or tree times with real printer and with virtual printer. The result have always been the same.  Ah ! I don't think that it could be related but I use the latest OctoPi.

kantlivelong, 2017-02-25:
Would you be able to do a test with timelapse on?

zegreatclan, 2017-02-25:
I'll check when my current print finishes, I keep you in touch ! Thank for your help. 

I don't know which event is thrown when the plugin log the "Printer is currently printing/paused" message. Maybe a bit of additional log here could help ? If Octoprint trigger the "done" event but the printer is still in "printing" state, it could be a OctoPrint issue.

kantlivelong, 2017-02-25:
That's what I'm guessing is happening but it seems weird that it would get PrintDone and still return True for is_printing() (Think I can safely assume that it wasn't paused)

So far I can't replicate it.

Are you running any plugins that would affect print functionality?

zegreatclan, 2017-02-25:
Here the additionnal plugins : DisplayProgress , EmailNotifier, PortLister, 
I'll try to uninstall them before more testing ;)

zegreatclan, 2017-02-25:
I made the tests :
I uninstalled all the plugins ... still not working,
I enabled timelapse, yep you are on something, it worked ! 

Still don't know why with a timelapse it works but ... in wait of a fix some timelapses won't kill me :D

gnosis953, 2017-02-26:
I also have the problem. Noting happens after print finishes.
So the solution might be to enable timelapse?

I will try that. thank you very much!

kantlivelong, 2017-02-26:
I wouldn't say its a solution but definitely a good workaround for now until I can figure out why it thinks a print is still running even though it's not.

gnosis953, 2017-02-26:
Works fine with timelapse enabled ;)

thank you for your plugin!
this solution will work for me the next time

kantlivelong, 2017-02-27:
I've committed a potential fix to the **issue-10** branch. Would you be willing to test?

#### How to install from branch
1.) Uninstall the plugin then install via URL.( https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/archive/issue-10.zip )

2.) Paste the URL into plugin manager(Example screenshot below):
[image]

gnosis953, 2017-02-27:
Thank you very much, i will test it tomorrow and write you if it does work

Thanks Steph

gnosis953, 2017-02-27:
> archive/issue-10.zip )

Hey, i did a short test.

I could not finish the print and i just canceled it manualy.
But the plug-in did nothing. The checkbox always changes it state.

The last time i canceled a print your plug-in also started the countdown to
shutdown.

Steph

Stephan Koch
E-Mail: steph.k933@gmail.com

2017-02-27 17:44 GMT+01:00 Stephan Koch <steph.k933@gmail.com>:

kantlivelong, 2017-02-27:
It shouldn't show the shut down timer if a job is cancelled but it will if the print failed.

gnosis953, 2017-02-27:
The new version starts countdown if i cancel the print and timelapse is
enabled.

Checkbox to shutdown changes its state sometimes.
That is the result of my research so far.

kantlivelong, 2017-02-27:
Hmm not entirely sure how I can rework this. If a job is cancelled OctoPrint will send PrintCancelled along with PrintFailed. PrintFailed can also be sent when there is a comm issue. I can check if the printer is an error state to determine failure vs cancel but where it gets tricky is when doing timelapse rendering because it will still render regardless if it was successful, cancelled, or failed and the plugin needs to wait for rendering to finish before shutting down.

Would be a lot easier if events sent a job id.

Will see if I can figure something out.

gnosis953, 2017-02-27:
I would be happy if it works without timelapse enabled. If countdown starts
when print gets canceled is no problem in my opinion.

I will although think about it. How it can be solved.

kantlivelong, 2017-03-01:
Alright I've reworked the timelapse support (which hopefully also fixed when not using timelapse). Should also allow jobs to be canceled without displaying the timer.

Same URL if you'd be willing to test it out. ( https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/archive/issue-10.zip )

Thanks

gnosis953, 2017-03-02:
Hey,

i tested it with timelapse enabled and disabled.

It works both times.

If print job is canceled countdown does not start.

Seems to work perfect in my opinion.

Steph

Stephan Koch
E-Mail: steph.k933@gmail.com

kantlivelong, 2017-03-03:
Awesome!

Merged into **devel**  ff82a3d5aaf73f9373e08349b87f77fc9996e6d6

commit 511ca875 (2017-03-06): Version 0.1.4

-Rework timelapse support (Fixes #10)
-Don't shutdown when a job is cancalled
commit 96271709 (2017-03-06): Version 0.1.4

-Rework timelapse support (Fixes #10)
-Don't shutdown when a job is cancelled
closed 2017-03-03 by kantlivelong