OctoPrint/OctoPrint-AutomaticShutdown #11: M80 / M81 Gcode on shutdown

macropin, 2017-03-10:
Is it possible to send `M81` on shutdown to actually power off the machine? There doesn't seem to be anyway to set this.

kantlivelong, 2017-03-10:
You could change your shutdown command to call a local script that first sends the GCode through the OctoPrint API then call the shutdown command.

kantlivelong, 2017-03-10:
You could also use an event handler

https://github.com/foosel/OctoPrint/wiki/Available-Events

foosel, 2017-03-10:
Better use http://docs.octoprint.org/en/master/events/index.html, event documentation is no longer maintained on the wiki :)

macropin, 2017-03-16:
Does the shutdown handler run before or after the server is ready to shutdown? If I send `M81` the power will be lost almost instantly which may not be a nice thing for the pi.

(As a side issue I'm looking at a way to add some hysteresis to the SSR that controls the power, so this shutdown race may be avoidable).

kantlivelong, 2017-03-16:
Anything done by OctoPrint would be before the shutdown process even begins on the system.

closed 2017-04-10 by kantlivelong