OctoPrint/OctoPrint-AutomaticShutdown #12: Auto shutdown during print

mikecarr, 2017-03-10:
If I select Auto Shutdown box during a print will it still execute at the end of the current print? It seems not to, just wanted to check. If not, is there a way to add that feature?

kantlivelong, 2017-03-10:
Yes it should.

mikecarr, 2017-03-10:
Ok, I will do some testing. At what point is the timer kicked off? Could it
be that it is waiting for timelapse to complete?

kantlivelong, 2017-03-10:
After the print finishes and timelapses are complete.

What version are on?

mikecarr, 2017-03-10:
Oh ok, that is probably why. I just updated to the latest version Octo:
1.3.1, Auto Shutdown: 0.1.4.

I will test out again, I may have been inpatient. What would be cool is if
there was an option to shutdown at different phases (i.e. after print
before timelapse).

Anyway, thanks for responding so quickly I really like this plugin. I have
it triggering a message to MQTT which powers off the lights and printer.

kantlivelong, 2017-03-10:
Why would you want to shutdown before the timelapse renders?

mikecarr, 2017-03-10:
I am not shutting down raspberry pi,  the printer does not need to be on
while it creates the video.

kantlivelong, 2017-03-10:
Oh. It's not really designed to turn off the printer PCB.

How are you shutting down the PCB now?

mikecarr, 2017-03-10:
I have some RF transmitters that are listening on a topic, it sends the
signal to Etekcity Wireless Remote controlled outlets which turn the power
on or off.

kantlivelong, 2017-03-10:
You could use an event to do this.

https://github.com/foosel/OctoPrint/wiki/Available-Events

mikecarr, 2017-03-10:
Nice! I will look into that.

foosel, 2017-03-10:
Better use http://docs.octoprint.org/en/master/events/index.html, event documentation is no longer maintained on the wiki :)

mikecarr, 2017-03-11:
I didn't even realize this, is there a way to find out if they are being
actively rendered or if it failed for some reason?

2017-03-11 05:56:04,918 - octoprint.plugins.automaticshutdown - INFO -
Waiting for 3 timelapse(s) to finish rendering before starting shutdown
timer...

kantlivelong, 2017-03-13:
Those could be from previous failed/aborted rendering jobs. You can clean them out from the **Timelapse Temp Folder**. 

They should automatically be cleaned up after 7 or so days though.

pdvorak57, 2017-03-18:
Is it possible to add an option to allow for automatically calling other system commands (for instance from https://github.com/Salandora/OctoPrint-SystemCommandEditor ) so that we can select to either power off the printer (with a wemo switch) or shutdown octopi, without having to modify the SystemShutdown command to the script to power off the printer instead of pi shutdown?

kantlivelong, 2017-03-20:
I'll add an option to specify a custom command as it has been requested a few time now.

Nappyy, 2017-10-27:
Hi there, I'm also waiting for a custom command.
Can you tell when a new version will be released?

kantlivelong, 2017-10-27:
Whoops! Forgot all about this.

It's in the devel branch if anyone would like to give it a spin and let me know how it works.

https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/archive/devel.zip

Nappyy, 2017-10-27:
I would like to try this version.
Unfortunately I don't know how to manually install this version.. :/

If you can give me a short description (or link) I would give it a try ;)

kantlivelong, 2017-10-27:
First uninstall the existing version. Then go to the dialog where you can install a plugin. Within that there will be a field to specify a URL. Use the URL in the previous post and click install.

Nappyy, 2017-10-27:
great!
Actually I have an active print. But after that (2-3hours) I'll give it a try :)

I'll give a reply this evening or tomorrow!

Nappyy, 2017-10-27:
After installing your updated plugin it's not possible to save any settings, because the "save" button is disabled and shows the "loading graphics (see attachment).
Same after restarting octoprint. After uninstalling the plugin it's working again.
[image]

kantlivelong, 2017-10-27:
Ah whoops. Try now but with this URL:
https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/archive/custom_command.zip

Nappyy, 2017-10-27:
Now save button is working fine.

I'm printing a ~45min job now.

Let's see if works :)

Nappyy, 2017-10-27:
It's working perfect!

Thanks for your quick support! :)

Nappyy, 2017-10-27:
Maybe it's possible to add another button in the "shutting down in... " popup -> "shut down now"

Also a small improvement could be to calc >60sec to minutes.
e.g.: "shutting down in 65" -> "shutting down in 1min 5sec"

kantlivelong, 2017-10-27:
please open a new ticket for the first or create a PR.

kantlivelong, 2017-12-13:
Closing this as the original query is resolved.

closed 2017-12-13 by kantlivelong