OctoPrint/OctoPrint-AutomaticShutdown #13: Automatic Shutdown Shuts Down Octoprint

Myko7, 2017-04-13:
The automatic shutdown function works, but it also shuts down Octoprint. I can only use Octoprint again after I restart the server. 

Is it possible to set the automatic shutdown function to activate a relay instead?

I currently use a relay board to control my printer's power. The following is a snippet from my Octoprint's config.yaml. 

system:
  actions:
  - action: pon
    command: printer_on
    name: Printer ON
  - action: poff
    command: printer_off
    confirm: Are you sure you want to turn off the printer?
    name: Printer OFF

I have tried changing the parameter in the plugin at several places below, but the plugin did not activate my relay board. Any suggestions?

**Line 92-94**
                if not self._settings.global_get(["system", "actions", "poff"]):
                        self._logger.warning("poff is not defined. Aborting shutdown...")
                        return

**Line 155-157**
def _shutdown_system(self):
		shutdown_command = self._settings.global_get(["system", "actions", "poff"])
		self._logger.info("Shutting down system with command: {action}".format(action=shutdown_command))

kantlivelong, 2017-04-13:
Seems like you want something like this instead:
https://github.com/kantlivelong/OctoPrint-PSUControl

Myko7, 2017-04-13:
Just curious. Do you know why automatic shutdown plugin doesn't work for my case? Thanks

P.S. I'm starting to learn python.

kantlivelong, 2017-04-13:
I don't think system actions parse the same way as regular settings. Never really looked. Someone in the OctoPrint IRC channel may be able to help.

closed 2017-05-12 by kantlivelong