OctoPrint/OctoPrint-AutomaticShutdown #14: adding one more button "shut down now"

Nappyy, 2017-10-27:
If you use a large abort timeout value it might be a good idea to add another button to "shut down now" in the popup (after the print job finished).
I think this would make it a bit more user-friedly in some cases.

hofcake, 2018-01-22:
I think the easiest way to do this is through the scripting that OctoPrint allows you to do. I hooked up a relay to the GPIO pins in my Pie and then added a button that would toggle the pins. I then had it enable/disable the pins when a print was started/ended.

alainm, 2019-01-30:
I was here just to add the same request: a manual **shutdown button**.
But let's add _another important functionality_: at the end of the print (and **cool down** + **extra time**) it could be configured to just shut down the printer. This can be done using separate scripts for PrinterOff and ShutDown and leaving the responsability to the user to configure those scripts according to his hardware.
This way it would cover a great range of needs :)