OctoPrint/OctoPrint-AutomaticShutdown #15: Does not shut down

hoegge, 2017-12-11:
Hi

I just installed the plugin but it does not shut down. I have checked the shutdown command is filled (and works). I have tried with different time out values - not clear what would abort timeout since nothing happens after a print.

kantlivelong, 2017-12-13:
Need more info.

- OS?
- OctoPrint Version?
- Shutdown command defined in OctoPrint?
- Include logs with the issue occurring

hoegge, 2017-12-18:
Hi, Sorry for delay

I updated to the latest version released last week or the week before and then it works. So cannot attach log with problem.

OctoPrint version : 1.3.6
OctoPi version    : 0.14.0
Linux octopi 4.9.28-v7+ #998 SMP Mon May 15 16:55:39 BST 2017 armv7l GNU/Linux
Command: sudo shutdown -h now

closed 2017-12-18 by kantlivelong