OctoPrint/OctoPrint-AutomaticShutdown #17: Add cool down delay before shutdown

StefanIGit, 2018-01-27:
My printer is connected by a master/slave  mains power socket thing, so when the octopi shuts down it cuts power to the printer.

Would be nice to be able to set hot-end and/or bed temp max limit before shutdown is executed.
Like set hot-end temp limit to 40°C so after the print the shutdown wait until the printer has cooled down the shutdown.

blandiant, 2018-04-17:
i agree, it would be nice

mikebzh44, 2018-04-25:
I agree too ;)

ThaliaFromPrussia, 2018-07-11:
I agree. Did you find the time to code it?

mikebzh44, 2018-07-11:
I have managed to turn off printer when hot end reach a desired T° by using GCode System commands plugin : 

https://plugins.octoprint.org/plugins/gcodesystemcommands/

So I create a OCTO1 command to power off the smart plug  : 

[image]

And I use it on my GCode end script.

yang3535, 2019-02-09:
Short of integrating with the g-code, a programmable cool down delay before shutdown would be nice.

aadithya96, 2020-07-03:
Wouldn't increasing the abort timer achieve the same requirement?

yang3535, 2020-07-03:
Not sure.  I generally don't like co-opting functions meant to serve a different scenario.  A shutdown is very different than an abort.  Would the abort timer apply?

gcadoret, 2021-08-03:
still open ? indeed, this is a requirement. it's not safe to shutdown printer when the tool is about 220°C. I thought it was managed , but too bad :'(

Klaws--, 2025-02-26:
> A shutdown is very different than an abort. Would the abort timer apply?

Well, the abort time (configured in the plugin's settings) is simply the time delay until the shutdown is performed. Nothing happens until that time runs out.

It's called "Abort Timeout" because Octoprint allows you to abort the shutdown during that time.

My printer reaches safe temperatures after 7 minutes, so my "Abort Timeout" is set to 300 seconds and the smart plug has another 120 second delay until it powers down.