OctoPrint/OctoPrint-AutomaticShutdown #26: Memory leak

mselkin, 2020-04-14:
Hi,

I believe I have discovered a memory leak in your plugin.
If you install your plugin and the Dashboard plugin (which allows you to view the raspberry Pi RAM usage), you will notice that every time you refresh the browser page, the RAM usage goes up by a significant amount and never recovers. The result is that if you remotely check on your print you eventually crash your raspberry Pi as it runs out of RAM.

I have systematically disabled and re-enabled every Plugin and I am certain that this Automatic shutdown plugin causes the browser to use up RAM without relinquishing it when the browser is closed.

I really depend on this plugin and would greatly appreciate any help or input you might have.

Thanking you,

Mark.

kantlivelong, 2020-04-28:
> I have systematically disabled and re-enabled every Plugin and I am certain that this Automatic shutdown plugin causes the browser to use up RAM without relinquishing it when the browser is closed.

> If you install your plugin and the Dashboard plugin (which allows you to view the raspberry Pi RAM usage), you will notice that every time you refresh the browser page, the RAM usage goes up by a significant amount and never recovers. 

Which is it? Does the leak occur with AutomaticShutdown by itself or does it require the Dashboard plugin to be installed as well?

mselkin, 2020-04-28:
I am not sure how to test that/answer your question because I rely on the Dashboard plugin to show me the Pi RAM usage.

If there is a way to see the current Pi RAM usage without the dashboard plugin, then I guess I could try that.

kantlivelong, 2020-04-28:
Turn it off and try to reproduce the issue reported. Doesn't need to be permanent. You can check ram utilization from the command line via ssh.

`free -mt`

mselkin, 2020-04-28:
Ok, I can try that.

I should mention, that if I turn off the checkmark in the automatic shutdown plugin that allows it to remember my setting of whether to auto shut down or not, then the memory leak does not occur.

In other words, I can have both Dashboard and Autoshutdown plugins installed and enabled and there is no memory leak, as long as that Autoshutdown plugin option is not checked.