OctoPrint/OctoPrint-AutomaticShutdown #27: Automatic Shutdown doesn't work anymore

gsxr2001, 2020-04-17:
Hello,
I have been using Automatic Shutdown for a long time and have never had any problems. For several days, it no longer works!

I tried both to delete the data and to reinstall the plugin and even the whole operating system!

I have OctoPrint 1.4.0 on OctoPi 0.17.0 and Raspberry Pi 3 B +

Thank you

Tenxbulls3y3, 2020-04-26:
I am new to Octoprint (version 1.4.0) and have the same issue with this plugin as gsxr2001. Installed it last  night, checked the box for Auto shutoff after print, did not work. Thanks for the help.
OctoPi Version 0.17.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

gsxr2001, 2020-04-26:
So could it be a compatibility issue with the latest versions of Octoprint?

kantlivelong, 2020-04-26:
Seems to work for me. Logs?

Tenxbulls3y3, 2020-04-26:
Hi Shawn,
New to this stuff. Is this what you are looking for”

If not, let me know what I need to set up under the Logging parameters in Octoprint and I will do so.
Thanks and regards,
Tony

kantlivelong, 2020-04-26:
https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

Just need a clean `octoprint.log` with a test print. Doesn't matter what the print job is. Can even be a few movements.

Tenxbulls3y3, 2020-04-26:
Here you go, this is the latest one

kantlivelong, 2020-04-26:
@Tenxbulls3y3 Attachments from email aren't supported by GitHub. Use a pastebin or file sharing.

Tenxbulls3y3, 2020-04-26:
[octoprint-2.log](https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/files/4536135/octoprint-2.log)
Sorry, here you go.

kantlivelong, 2020-04-26:
There are some very strange gaps in that logfile. But I do see the shutdown event being triggered

```
2020-04-26 08:11:06,949 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2020-04-26 08:11:06,963 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: CE3_ear_safe_fast_strap_mask.gcode, owner: TenXBulls3y3
2020-04-26 08:11:07,037 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2020-04-26 08:11:07,059 - octoprint.plugins.automaticshutdown - INFO - Starting abort shutdown timer.
2020-04-26 08:11:07,745 - octoprint.plugins.tracking - INFO - Sent tracking event print_done, payload: {'origin': u'local', 'file': '31a5409406f8c18dd097b894bd83f17ba068ca9a', u'elapsed': 16128}
2020-04-26 08:11:37,085 - octoprint.plugins.automaticshutdown - INFO - Shutting down system with command: sudo shutdown -h now
2020-04-26 08:11:37,688 - octoprint.server - INFO - Shutting down...
2020-04-26 08:11:38,403 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-04-26 08:11:38,407 - octoprint.events - INFO - Event loop shut down
2020-04-26 08:11:38,604 - octoprint.server - INFO - Goodbye!
```

gsxr2001, 2020-04-28:
This is my log
[octoprint.log.2020-04-14.txt](https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/files/4547613/octoprint.log.2020-04-14.txt)

kantlivelong, 2020-04-28:
@gsxr2001 which timeframe should a shutdown timer have been presented?

gsxr2001, 2020-04-28:
> @gsxr2001 which timeframe should a shutdown timer have been presented?

30

kantlivelong, 2020-04-28:
@gsxr2001 When did you expect to see the timer or a shutdown event?

gsxr2001, 2020-04-28:
> @gsxr2001 When did you expect to see the timer or a shutdown event?

when printing ends

kantlivelong, 2020-04-28:
Based on what provided I see
```
2020-04-12 18:18:36,307 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2020-04-12 18:18:36,613 - octoprint.util.commandline - ERROR - Error while trying to run command /usr/bin/avconv -framerate 25 -i "/home/pi/.octoprint/timelapse/tmp/GA30_Screw_Side_20200412123544-%d.jpg" -vcodec mpeg2video -threads 1 -r 25 -y -b 10000k -f vob -vf '[in] format=yuv420p [postprocessed]; movie=/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/static/img/watermark.png [wm]; [postprocessed][wm] overlay=10:main_h-overlay_h-10 [out]' "/home/pi/.octoprint/timelapse/.GA30_Screw_Side_20200412123544.mpg"
2020-04-12 18:18:36,615 - octoprint.timelapse - WARNING - Could not render movie, got return code None: []
2020-04-12 18:18:41,315 - octoprint.plugins.telegram - INFO - Sending a message: Stampa di GA30_Screw_Side.gcode terminata! with_image=True with_gif=False chatID= 371560499
2020-04-12 18:18:41,438 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
......
2020-04-12 18:48:17,846 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
2020-04-12 18:48:22,971 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
2020-04-12 18:48:28,104 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
2020-04-12 18:48:31,100 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.86
2020-04-12 18:48:31,973 - octoprint.server.util.sockjs - INFO - User fdl logged in on the socket from client ::ffff:192.168.1.86
2020-04-12 18:48:33,249 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
2020-04-12 18:48:35,677 - octoprint.server.api.system - INFO - Performing command for core:shutdown: sudo shutdown -h now
2020-04-12 18:48:36,056 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.86
2020-04-12 18:48:36,325 - octoprint.server - INFO - Shutting down...
2020-04-12 18:48:37,327 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-04-12 18:48:37,331 - octoprint.events - INFO - Event loop shut down
2020-04-12 18:48:37,354 - octoprint.plugins.telegram - INFO - An Exception in get final time : unsupported type for timedelta seconds component: unicode
2020-04-12 18:48:37,371 - octoprint.server - INFO - Goodbye!
```

Which indicates that timelapses were being processed which prevents the shutdown timer from starting. Otherwise it looks fine. If that is not the case please provide FRESH logs and detailed information.

gsxr2001, 2020-04-28:
> Based on whats been provided I see
> 
> ```
> 2020-04-12 18:18:36,307 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
> 2020-04-12 18:18:36,613 - octoprint.util.commandline - ERROR - Error while trying to run command /usr/bin/avconv -framerate 25 -i "/home/pi/.octoprint/timelapse/tmp/GA30_Screw_Side_20200412123544-%d.jpg" -vcodec mpeg2video -threads 1 -r 25 -y -b 10000k -f vob -vf '[in] format=yuv420p [postprocessed]; movie=/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/static/img/watermark.png [wm]; [postprocessed][wm] overlay=10:main_h-overlay_h-10 [out]' "/home/pi/.octoprint/timelapse/.GA30_Screw_Side_20200412123544.mpg"
> 2020-04-12 18:18:36,615 - octoprint.timelapse - WARNING - Could not render movie, got return code None: []
> 2020-04-12 18:18:41,315 - octoprint.plugins.telegram - INFO - Sending a message: Stampa di GA30_Screw_Side.gcode terminata! with_image=True with_gif=False chatID= 371560499
> 2020-04-12 18:18:41,438 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
> ......
> 2020-04-12 18:48:17,846 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
> 2020-04-12 18:48:22,971 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
> 2020-04-12 18:48:28,104 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
> 2020-04-12 18:48:31,100 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.86
> 2020-04-12 18:48:31,973 - octoprint.server.util.sockjs - INFO - User fdl logged in on the socket from client ::ffff:192.168.1.86
> 2020-04-12 18:48:33,249 - octoprint.plugins.automaticshutdown - INFO - Waiting for 2 timelapse(s) to finish rendering before starting shutdown timer...
> 2020-04-12 18:48:35,677 - octoprint.server.api.system - INFO - Performing command for core:shutdown: sudo shutdown -h now
> 2020-04-12 18:48:36,056 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.86
> 2020-04-12 18:48:36,325 - octoprint.server - INFO - Shutting down...
> 2020-04-12 18:48:37,327 - octoprint.events - INFO - Processing shutdown event, this will be our last event
> 2020-04-12 18:48:37,331 - octoprint.events - INFO - Event loop shut down
> 2020-04-12 18:48:37,354 - octoprint.plugins.telegram - INFO - An Exception in get final time : unsupported type for timedelta seconds component: unicode
> 2020-04-12 18:48:37,371 - octoprint.server - INFO - Goodbye!
> ```
> 
> Which indicates that timelapses were being processed which prevents the shutdown timer from starting. Otherwise it looks fine. If that is not the case please provide **FRESH** logs and detailed information.

AS doesn't work with active timelapse?

kantlivelong, 2020-04-28:
@gsxr2001 It waits for the timelapse to finish processing otherwise your timelapse video would not be generated.

gsxr2001, 2020-04-28:
> @gsxr2001 It waits for the timelapse to finish processing otherwise your timelapse video would not be generated.

ok, but after it finishes processing, it should turn off

kantlivelong, 2020-04-28:
Correct. It will display the dialog after timelapses are finished processing.

gsxr2001, 2020-04-28:
> Correct. It will display the dialog after timelapses are finished processing.

But it doesn't work anymore for me

kantlivelong, 2020-04-28:
@gsxr2001 Try installing [this](https://github.com/OctoPrint/OctoPrint-AutomaticShutdown/archive/d8bca82bdc7ca40672c330807c2411f99dde61a4.zip) version and generate fresh logs. It is little more verbose which should help pinpoint the issue.

gsxr2001, 2020-04-29:
I tried to disable the timelapse and AS works again