OctoPrint/OctoPrint-AutomaticShutdown #35: Does not shutdown after Timelapse has finised

HenrikDK1971, 2023-09-10:
The Automatic shutdown works quite well until I turn on timelapse, - then it waits for the timelaps which is fair. But after a fine notification comes it still does not tune octoprint off, - is this a general problem?

theFivehead, 2024-07-06:
I think it is because I have exactly the same problem.
This makes this plugin almost useless. 😞

Klaws--, 2025-02-26:
In my case, it seems to work only when the Octoprint GUI is being displayed in a web browser.

My typical procedure is to start the print (via Cura or the Octoprint GUI), then I tick the "Shutdown after print finished" checkbox, then I close the web browser with the Octoprint GUI (and shut down the computer as well), and then I wait.

Octoprint doesn't shut down after the print finished.

When I open the web browser with the Octoprint GUI, the message box appears which allows me to abort the shutdown and Octoprint shuts down after some time (abort timeout is set to 300 seconds), unless (obviously, I click the abort button (because I want to start the next print).